thomast777/grbl32 #2: Spindle/Laser PWM Frequency output pin

RCSrohan, 2019-09-06:
Hi there Tom, 

Firstly many thanks for your awesome coding and support with grbl32, its great. 

So this is merely a suggestion regather than an issue, 

Is there any way of coding in a PWM output frequency for the spindle/laser PWM output pin, 

Most laser drivers run at different frequencies and I know mine runs best at 50khz. 

My laser PSU is a K40 = MYJG40W

TH | Input Signal | On-Off laser control,TH≥3V, emitting laser; TL≤0.3V, no laser.
-- | -- | --
TL | Input Signal | On-Off laser control,TH≥3V, no laser; TL≤0.3V, emitting laser
WP | Input Signal | On-Off laser control,TH≥3V, no laser; TL≤0.3V, emitting laser
G | GND | This foot must be connected well with the laser machine shell and the ground of control board.
IN | Input Signal | The control of laser power: Both 0-5V analog signal and 5V PWM signal can control the laser power.
5V | Output Power | Output 5V, the maximum output current is 20mA.

H | Input Signal | On-Off laser control,TH≥3V, emitting laser; TL≤0.3V, no laser.
-- | -- | --
TL | Input Signal | On-Off laser control,TH≥3V, no laser; TL≤0.3V, emitting laser
WP | Input Signal | On-Off laser control,TH≥3V, no laser; TL≤0.3V, emitting laser
G | GND | This foot must be connected well with the laser machine shell and the ground of control board.
IN | Input Signal | The control of laser power: Both 0-5V analog signal and 5V PWM signal can control the laser power.
5V | Output Power | Output 5V, the maximum output current is 20mA.

So the real question is whether or not we can include a grbl $$ setting to control the PWM frequency output. 

For information on these laser systems go here 

http://donsthings.blogspot.com/2016/06/k40-s-laser-power-control-interface_86.html?m=1

thomast777, 2019-09-09:
Hello Don,

At the moment, I am adding the mod using the $40 range settings to change the PWM frequencies. This was needed for the extended number of PWM channels on the F46 version, something which is hard-coded now for 8 RC servos. 

In addition, a $50 configuration to enable/invert the spindle_enable pin while in laser mode.

I should have these up in about a week if things are tested out ok on my system.

mike-pittelko, 2020-01-20:
Would be great to have this added to the documentation.