thomast777/grbl32 #3: grbl1.1h

mnesarco, 2019-10-25:
Hi @thomast777 , great job! STM32 is an amazing platform and probably the future of grbl.

One question, grbl1.1h was released with some bugfixes, do you plan to keep this project in sync with gnea/grbl? I can see that this repo is not a fork of gnea/grbl so i wonder how will you maintain it up to date with mainstream.

mnesarco, 2019-10-26:
I did'n realized that this repo is based on the grbl_Mega and not on grbl, there is no 1.1h yet on grbl_Mega repo :(

I have been reading the code and I don't get the point of keeping ATMEGA328P specific code, this project can be maintained much better if you drop all the ATMEGA328P specific code.

thomast777, 2019-10-26:
I am looking at the bug fixes now, and will see if I can make this branch "1.1h". 

The reason for the ATMEGA328P code is only to show the locations where the STM32 code was needed, more or less for educational purposes to highlight where GRBL hits the metal, and the difference in low-level hardware calls (GRBL32 was never meant to be dual architecture). 

Perhaps the ATMEGA code will be shown as comments, just to show GRBL32 roots.

lianzaozi, 2019-11-27:
Hello, do you have any specific requirements for the hardware used with grbl32? Can it be used with ordinary hardware? Is there any way to buy off-the-shelf hardware?

cbcy1234, 2020-03-21:
error: expected declaration specifiers or '...' before '(' token  
6 axis can't be compiled, please ask for some thanks!

closed 2024-10-30 by mnesarco