thomast777/grbl32 #7: F13 bluepill grbl 32 controller board keeps disconnecting

RCSrohan, 2020-03-02:
Anyone have any thoughts as to why my controller board keeps disconnecting from usb com port, 

So I connect via lasergrbl and all is well, but every now and again when a job is running the board disconnects from my laptop and the job is then useless. 

I've tried so many different drivers. And am now looking into a fresh install of Windows just to rule them out but could I have a defective board ???

thomast777, 2020-03-02:
Hello RCSrohan,

Please send your purchase information to support (at) tomsrobotics.com. I
will arrange to send out a replacement for you to try.

Best,
Tom

cbcy1234, 2020-03-21:
error: expected declaration specifiers or '...' before '(' token

GIANNHSitia, 2020-04-18:
I have the same problem, after a while it crashes and you lose the connection to the computer. my board is homemade. I testing without connect stepper motor and spindle and is same. mabe have problem the code. I have connect with ttl board to B6 B7 and run with candle1.2.b.1. with UGS running better

qiwalker, 2020-04-21:
hi .. any tutorial how to flash to STM32F103 board ?

GIANNHSitia, 2020-04-21:
Yes. I can send you a bin file and flash with stm32 flash loader.

qiwalker, 2020-04-21:
@GIANNHSitia thank you ^_^ please send to my mail, rizki.walker@gmail.com

can i use arduino IDE to flash grbl32 firmware (from github) to SMT32 board ?

GIANNHSitia, 2020-04-21:
hello again. I send you email.

qiwalker, 2020-04-22:
hello thanks, btw did u already solve ur problem about lose connection ?

GIANNHSitia, 2020-04-22:
With Universal Gcode Sender work ok. But with candle after few moment freeze the connection

e-zeki, 2020-05-13:
> 
> 
> Yes. I can send you a bin file and flash with stm32 flash loader.

@GIANNHSitia  Hello GIANNHSitia, Can you help me out how to run this code?I justhave a few questions? would be mean tons of help for me  if you can. please contact with me on solakzeki@gmail.com
Zeki
Thanks

wandlerdarwin, 2023-08-10:
USB cable is too long and not shielded and grounded I use to have the same issue had to breakdown and find good cable and pay what it cost to eliminate error

wandlerdarwin, 2023-08-10:
also you have to turn off all the windows garbage that is pushing interrupts glitching your code. I have no network connection and use a portable usb wifi jack if I need connection then I unplug it. Firewall off all protection off  all updates junk off all security off all virus programs I have removed from computer cause windows hides crap I have had to go into settings apps and search for junk and remove it from my computer  hope it helps