thomast777/grbl32 #8: 6 axis can't be compiled, please ask for some thanks!

cbcy1234, 2020-03-21:
error: expected declaration specifiers or '...' before '(' token

cbcy1234, 2020-03-21:
6轴 无法编译通过 麻烦请教一些  谢谢！

cbcy1234, 2020-03-21:
6 axis can't be compiled. Please ask for help. Thank you!

mvwtest, 2020-04-25:
Me too.
stm32f103 / 6 axis /.Atollic/F16  can't be compiled
error: expected declaration specifiers or '...' before '(' token
do you  resolve it？
Any suggest.thanks

MrKKrolik, 2020-04-28:
/stm32f407xx.h:1058:29: error: expected declaration specifiers or '...' before '(' token
have the same trouble in stm32f407VET6 ((

MrKKrolik, 2020-05-08:
Problem on the VARIABLE_SPINDLE_ENABLE_PIN, without it everything building fine

samitray, 2020-05-16:
@MrKKrolik  were you able to find a work around to the compile issue

samitray, 2020-05-16:
I thin it may be due to macro clashing issue because of  double definiation
of Spindle_Disable() and Spindle_Enable() in stm32_pin_out.h and stm32utilities.h
So if macro define in stm32_pin_out.h is commented the project builds.... havent checked if it works yet

neuronthc, 2020-12-01:
Same, macro #define Spindle_Disable() LL_TIM_CC_DisableChannel(SPINDLE_TIMER, SPINDLE_CHANNEL) 
and #define Spindle_Enable() LL_TIM_CC_EnableChannel(SPINDLE_TIMER, SPINDLE_CHANNEL) in the stm32_pin_out.h
are conflicted with functions with same signature.
void Spindle_Disable() and void Spindle_Enable()

wlzxzq, 2022-09-23:
> pindle_Disable() 和无效 Spindle_Enable()

can you share your all code for stmf4 with six axis project

codehat, 2022-11-29:
Just comment Spindle_Disable and Spindle_Enable macro definitions in file  stm32\stm32_pin_out.h will resolve this compile error.
But I do not test any function about spindle, any one tried please share your result.