thomast777/grbl32 #11: Compiled STM32CubeIDE F13 on blue pill SERIAL USB not detected.

AxelGUILLAUMET, 2020-05-06:
Hi!

This actualized STM32-GRBL project looks to be very good but I have some issues...

I recompiled the project successfully with STM32CubeIDE, I load the firmware on the blue pill and there is nothing detected by the PC.

I had in the past the same issue and iw was due to a bad clone of the STM32F103.

But this time it can't be the STM32F103 I have on my blue pill because when I compile another firmware, it works (This one -> https://github.com/ewidance/grbl-usbcnc-stm32) 

I send you there all the specs of my hardware...

The .elf file of the F13 compiled on STM32CubeIDE

And the working .elf file I compiled form https://github.com/ewidance/grbl-usbcnc-stm32 which is working

I think it can be  an USB activation issue in the source code but I don't found nothing clear about that...

[image]

[image]

[STM32 compiled firmwares.zip](https://github.com/thomast777/grbl32/files/4584382/STM32.compiled.firmwares.zip)

mwshakoor154, 2020-05-18:
i think this only supports USART, no usb support.
I also tested, usb not detected.
as far as my understandings after checking hardware notes, it clearly shows PB6 = Tx, PB7= RX.
https://github.com/thomast777/grbl32/wiki/Hardware-Notes
I have not tested using USART but i guess it will work.

mwshakoor154, 2020-05-19:
Today I checked f13.
UART is working
set baud rate to 921600

AxelGUILLAUMET, 2020-05-20:
OK thanks a lot, I confirm this, if you are looking at this page which is the page of the oficial hardware of this firmware, [](https://shop.tomsrobotics.com/product/f13-bundles/) you can see clearly the 12Mhz crystal, used by the CH340 USB to Serial chip. 

Here in the F16 https://shop.tomsrobotics.com/product/f16-bundles/ you cna see clearly the CH340.

Tomorrow I will make a test with real hardware. I plan to use it with LightBurn and a CO2 Synrad RF laser of 40 Watts.

AxelGUILLAUMET, 2020-05-21:
OK USART confirmed and lightburn GRBL compatibility is OK

[image]

[image]

I modded for the moment src/usart.c to change the speed to 115200bps due to the FT232 module is a counterfeit and is only limited to 115200 Max. But to test and validate this GRBL port on STM32 platform it is the best.

Thanks and regards.

closed 2020-05-21 by AxelGUILLAUMET