thomast777/grbl32 #15: UART F46

IhsanFuadi3, 2020-12-01:
Hi,i tried to connect with UGS Platform,but it didn't work
can you help me?
FYI,i use GRBL F46

thomast777, 2020-12-01:
Hi,

By default, the F46 is running at 921,600 baud rate. You can enter this
value in the baudrate setting of UGS then press enter, then try to connect.

Best,
Tom

IhsanFuadi3, 2020-12-01:
thanks for the reply sir,i've tried your advice,but didn't work,grbl comunication using usart?

anankervis, 2020-12-01:
My F46 also had trouble connecting. I tracked it down to the serial -> USB
module. I bought an alternate, plugged it into different pins, and
configured the board to use it instead.

anankervis, 2020-12-02:
> On Tue, Dec 1, 2020 at 7:25 AM IhsanFuadi3 *@*.***> wrote: thanks for the
> Are you using serial ch340? Do you use different USART pin?

Hi, it's this:
https://www.amazon.com/gp/product/B07TS3GPQ1
with an FTDI FT232RL.

I connected it to the Comm RX TX and G pins. It's been a while since I set
this up, I believe those pins are electrically connected to the same I/O
pins that the built-in USB -> TTL chip uses.

IhsanFuadi3, 2020-12-02:
oke sir,thanks 
i will try again

muhkusen, 2020-12-02:
> My F46 also had trouble connecting. I tracked it down to the serial -> USB module. I bought an alternate, plugged it into different pins, and configured the board to use it instead.
> […](#)

Hi Mr.  Anankervis, What the different pin do you use?

anankervis, 2020-12-02:
I didn't use a different pin after all - I mis-remembered. I just used the
Comm RX TX G pins. I believe those are the same IO pins used by the
built-in chip, so you shouldn't need to re-configure anything.

muhkusen, 2020-12-02:
> I didn't use a different pin after all - I mis-remembered. I just used the Comm RX TX G pins. I believe those are the same IO pins used by the built-in chip, so you shouldn't need to re-configure anything. On Tue, Dec 1, 2020 at 8:59 PM Muhamad Kusen <notifications@github.com> wrote:
> […](#)

ok thanks for your respon, may I know what the board do you use ?

anankervis, 2020-12-02:
The 6-axis F46. I used this to connect to it:
https://www.amazon.com/gp/product/B07TS3GPQ1
with an FTDI FT232RL chip.

wlzxzq, 2022-09-23:
> 6 轴 F46。我用它来连接它： [https](https://www.amazon.com/gp/product/B07TS3GPQ1) : //www.amazon.com/gp/product/B07TS3GPQ1 和 FTDI FT232RL 芯片。2020 年 12 月 1 日星期二晚上 11:29 Muhamad Kusen <notifications@github.com> 写道：
> […](#)
can you  share your f4 project?

WolfgangAipperspach, 2022-10-24:
Hi Thomas,
is the following the correct serial setting for the F46?
* 921600 baud
* 8 data bits
* No parity bit
* 1 stop bit
I can get cummunication but strange symbols only

WolfgangAipperspach, 2022-10-24:
Seems to work with 617284baud 8N1 on F46. I found this measuring the duration of one bit on an oscilloscope (~1.62 us). It does not seem to be a standard baud rate.

wlzxzq, 2022-12-14:
you can test 115200, modify the source code